donghoonpark/ltspice_pytool #1: "unpack requires a buffer" issue

DMWilken, 2019-09-03:
Hello,
I wanted to use the ltspice package but experience an issue. Parsing a *.raw file I get the error message: _unpack requires a buffer of 26020 bytes_

I run the following lines:
```
import ltspice
import matplotlib.pyplot as plt
import numpy as np
import os
path = os.getcwd()+'\\test.raw'
l = ltspice.Ltspice(path) 
l.parse() 
```

and get the error

```
---------------------------------------------------------------------------
error                                     Traceback (most recent call last)
<ipython-input-20-c03622bc666e> in <module>()
      5 path = os.getcwd()+'\\test.raw'
      6 l = ltspice.Ltspice(path)
----> 7 l.parse()

~\AppData\Local\Programs\Python\Python37\lib\site-packages\ltspice\ltspice.py in parse(self, dsamp)
     77             self.t_list.append(vdata[2])
     78 
---> 79         self.data_raw = struct.unpack(str(self.p_number*(self.v_number+1))+'f', data[i:size])
     80 
     81         self.time_raw = [None]*self.p_number

error: unpack requires a buffer of 26020 bytes

```

Depending on my simulation configuration the buffer size changes.

My test code was

```
c1 gnd n$1 1p
v1 n$1 GND DC 0 AC 5 0 sin(0 {1.4142*5} 50 {-(0+360)/360/50} 0) rser=0
.AC dec 1k 10Meg 100G
.END
```

An example *.raw file can be found [here](https://wolke7.aei.mpg.de/s/P35wJQgK3gFEGse).

I am using python version 3.7.150.1013.

Thanks for your help!

DMWilken, 2019-09-04:
Forgot to mention that I am using LTspiceXVII. I just read that this version uses 16bit Unicode while LTspiceIV used ANSII for the raw files. Maybe this something to do with the issue.

DongHoonPark, 2019-09-17:
Is the problem occur in all .raw files?

DongHoonPark, 2019-12-10:
Cause there is no response from @DMWilken , I'll close this issue. If anyone is suffering that kind of issue, please report me via any channel.

closed 2019-12-10 by DongHoonPark

KaoutarHazim, 2020-05-01:
I am actually having the same error.

KaoutarHazim, 2020-05-01:
> Cause there is no response from @DMWilken , I'll close this issue. If anyone is suffering that kind of issue, please report me via any channel.

Hello, I am actually having the same error..

DongHoonPark, 2020-05-19:
@KaoutarHazim Due to the issue, from the commit [#710646b ](https://github.com/DongHoonPark/ltspice_pytool/commit/710646b0d95ceb9f58e841a54c36b1841e2fc52c), the struct unpack operation for whole buffer alternated with numpy function. The only remained part of struct usage is the rebuilding 64bit floating point value of time. So, i thought that the issue had been resolved. If you use older version of my library, please kindly update it. Or, if you are facing the problem even if you use new version, please open another issue and give me the signal file so that i can debug it.