donghoonpark/ltspice_pytool #2: Add support for .fft files

the-moog, 2020-01-02:
FFT simulation results (.fft) files have a very similar format to transient and ac simulation (.raw) files, but attempting to load one creates an error.
Having support for .fft files would be desireable.

`---------------------------------------------------------------------------
error                                     Traceback (most recent call last)
<ipython-input-15-c512f1c7b7e1> in <module>
----> 1 lts.parse()

c:\python37-64\lib\site-packages\ltspice\ltspice.py in parse(self, dsamp)
     77             self.t_list.append(vdata[2])
     78 
---> 79         self.data_raw = struct.unpack(str(self.p_number*(self.v_number+1))+'f', data[i:size])
     80 
     81         self.time_raw = [None]*self.p_number

error: unpack requires a buffer of 2621420 bytes
`

the-moog, 2020-01-02:
I've cleaned the code as per PEP8 and added code to handle .fft files, see pull request

DongHoonPark, 2020-03-30:
I merged your code and did some refactoring. Now FFT & AC analysis file can be read.

closed 2020-03-30 by DongHoonPark