donghoonpark/ltspice_pytool #4: Mismatch in data read from.raw by ltspice Python class and LtSpice itself

PetrShepelev, 2020-05-05:
Here is zip file attached with a test case waveforms. Having this .raw file loaded in LtSpice (XVII x64 Feb25 2020) show the peak values reached by signal I(R0_6): -1.6mA at 1ns time and 1.6mA at 502ns.
Having the same file loaded and parsed with ltspice Python class, the min and max values of the signal are -8.909788448363543e-05 and 8.913704368751496e-05 correspondingly.
Note, the same LtSpice version was used to run the simulation generated this .raw file.
[test20200505_15_59_47.zip](https://github.com/DongHoonPark/ltspice_pytool/files/4584109/test20200505_15_59_47.zip)

DongHoonPark, 2020-05-16:
I'm afraid that cannot reproduce your error situation.. 
With the code below, I can get the proper result. 
`
[In]
import ltspice
import numpy as np

l = ltspice.Ltspice('test20200505_15_59_47.raw')
l.parse()

print(np.max(l.getData('I(R0_6)')))

[Out]
0.0016032476
`
If you don't mind, give me more specific situation and your system setup.

PetrShepelev, 2020-05-19:
I looked deeper into the issue, and recreating it in an isolated code from my main development.
Here I have attached the code  and two raw files reading those in sequence allow me to recreate the problem i have had initially.
 
As a side note, I believe the reason for the problem is because of class Ltspice has members data_raw and time_raw declared as class variables rather than instance variables. Anyways, here is the issue, note min and max data from test_f.raw is different depending if test_b.raw was parsed before it
[Test1_2.zip](https://github.com/DongHoonPark/ltspice_pytool/files/4653775/Test1_2.zip)

:
[Test1 script]
import ltspice

def run_test1():
    l_b = ltspice.Ltspice('test_b.raw')
    l_b.parse()
    test_b = l_b.getData('I(R0_6)')
    print(f"test_b  min={test_b.min()}, max={test_b.max()}")

    l_f = ltspice.Ltspice('test_f.raw')
    l_f.parse()
    test_f = l_f.getData('I(R0_6)')
    print(f"test_f  min={test_f.min()}, max={test_f.max()}\n")

if __name__ == '__main__': run_test1()
[Out]
test_b  min=-0.00160311465151608, max=0.0016032606363296509
test_f  min=-8.908558811526746e-05, max=8.916806109482422e-05

[Test2 script]
import ltspice

def run_test2():
    l_f = ltspice.Ltspice('test_f.raw')
    l_f.parse()
    test_f = l_f.getData('I(R0_6)')
    print(f"test_f  min={test_f.min()}, max={test_f.max()}\n")

if __name__ == '__main__': run_test2()
[Out]
test_f  min=-0.0016031228005886078, max=0.001603269367478788
[two_raw_files.zip](https://github.com/DongHoonPark/ltspice_pytool/files/4653771/two_raw_files.zip)

DongHoonPark, 2020-05-21:
Oh.. It also is an malfunction derived by my mistake that same as #5 .
I already recognized the problem and changed that class variables to instance variables on commit [c29ade8](https://github.com/DongHoonPark/ltspice_pytool/commit/c29ade87da26411fc430c76e3d7a1299d47ff334)
I also updated pip version (0.4.2) 
If you use latest pip version, the output for the Test1 script will like below.

```
test_b  min=-0.00160311465151608, max=0.0016032606363296509
test_f  min=-0.0016031228005886078, max=0.001603269367478788
```

Thank you for your report and if you have any other problem, please feel free to make issue.

closed 2020-05-21 by DongHoonPark

PetrShepelev, 2020-05-21:
Thanks for the fix.

harivignesh1608, 2021-09-29:
Reading time data from a raw file gives negative values for time data. While the same data plotted and exported as text file in Ltspice looks fine. 
I have attached all the necessary files in the zip which is a simple RC circuit.
I hope I have given the necessary details about the bug.
[Draft3.zip](https://github.com/DongHoonPark/ltspice_pytool/files/7251987/Draft3.zip)
.

DongHoonPark, 2021-09-29:
That bug was fixed in version 1.0.5, which I released just a minute ago.
In addition to bug fixes, I included a test component to detect this type
of issue.
Thank you for your patience, and please try version 1.0.5.
Regards.

harivignesh1608, 2021-10-02:
Thanks for the bugfix.