donghoonpark/ltspice_pytool #5: Data from multiple parse() operations gets appended to an ltspice object instead of overwriting

dextersLabRat, 2020-05-14:
Hi. I'm using your lib in a Jupyter Notebook so that you have some context. I have also installed ltspice via pip in a Conda environment (the environment that my notebook is running in).

When I read in a raw file into a variable/object using:
```
l = ltspice.Ltspice(filepath)
l.parse()
```
the first time after (re)loading the Notebook kernel, there is no problem. However, even if I delete the 'l' variable created before, any new invocations like the above _append_ the data to the 'l' variable instead of creating a new object. The only way to start with a fresh object it seems is to reload the kernel.

Using a circuit identical to your '01-RC Circuit' example, here is my code:
```
lts = ltspice.Ltspice(filepath)
lts.parse() # Data loading sequence. It may take few minutes.

lts.getVariableNames()
```
And the output:
```
['time', 'V(source)', 'V(cap)', 'I(C1)', 'I(R1)', 'I(V1)']
```
So far, so good. I then do another iteration of the above and get as output:
```
['time',
 'V(source)',
 'V(cap)',
 'I(C1)',
 'I(R1)',
 'I(V1)',
 'time',
 'V(source)',
 'V(cap)',
 'I(C1)',
 'I(R1)',
 'I(V1)']
```
That is, now everything has been duplicated.

Here is another example of two iterations in one notebook cell after reloading the kernel:

```
lts = ltspice.Ltspice(filepath)
lts.parse() # Data loading sequence. It may take few minutes.

display(lts.getVariableNames())

ts = ltspice.Ltspice(filepath)
lts.parse() # Data loading sequence. It may take few minutes.

display(lts.getVariableNames())
```

Output:
```
['time', 'V(source)', 'V(cap)', 'I(C1)', 'I(R1)', 'I(V1)']
['time',
 'V(source)',
 'V(cap)',
 'I(C1)',
 'I(R1)',
 'I(V1)',
 'time',
 'V(source)',
 'V(cap)',
 'I(C1)',
 'I(R1)',
 'I(V1)']
```

The first display() call shows expected behavior, but the second shows the data being appended rather than getting overwritten (as I would expect, given that I've reassigned the 'lts' variable).

I also note that if I use two variable names, that doesn't change the appending behavior. The only thing that gives me the expected behavior is doing ltspice.Ltspice() and parse() calls after reloading the kernel. But only that first time. Thanks.

DongHoonPark, 2020-05-16:
Thank you for your detailed bug report! 

The problem have derived from the reason that library shared single list instance which is created when library loading. So I changed the structure simply to create all of the lists used in library  within the ltspice instance initialization process. 

You can use patched version either on github and pypi (0.4.2).

closed 2020-05-16 by DongHoonPark

dextersLabRat, 2020-05-18:
Awesome! I'll try it out! Thanks for the quick response. I'm glad my attempt at a bug report was helpful. :)