donghoonpark/ltspice_pytool #10: How to read back value of stepped parameter when case_count > 1

abergerSRS, 2021-01-20:
It would be useful if it was possible to query the value of a stepped parameter for simulations where case_count > 1. For example, in your rectifier.asc, Cdc is stepped through various values, which could be useful for providing a legend to the plot of V_cap vs. time.

DongHoonPark, 2021-01-28:
I absolutely agree with your opinion, however, the information of step parameter is recorded in the .asc file, not the .raw file. Indeed, we need to bring an auxiliary .asc file to achieve the function you mentioned. 
I may add this function in the form of connection of .raw and .asc file but it could be a little bit complicated. 
Anyway, thank you for your opinion and I will consider it. If you have any great ideas for implementation, please let me know.
Regards.

closed 2021-01-28 by DongHoonPark

abergerSRS, 2021-01-28:
I discovered that the .log file associated with the .asc file contains information about the stepped parameter. The .log file is easy to read in and parse with Python. For example:

```
with open('rc_lowpass_step_c.log', 'r') as logFile:
    cases = []
    for line in logFile.readlines():
        if(line.startswith('.step')):
            cases.append(line.strip('.step [').rstrip())]
```
[rc_lowpass_step_c.log](https://github.com/DongHoonPark/ltspice_pytool/files/5888607/rc_lowpass_step_c.log)

DongHoonPark, 2021-02-02:
As you mentioned, it requires to read other file. and if there is 2< stepped parameter or variables that require calculation, it will be more complicated. 
I'm now working on code to extract test values in .asc so, please keep your eyes to my repo. 
Regards.