donghoonpark/ltspice_pytool #14: ltspice module fails to handle if No. of variables greater than 2000

Goutham-AKA, 2021-03-04:
Dear Dong,

While I was using the ltspice module to perform my circuit simulation in python which has a number of variables greater than 2000 in raw file, the module fails to load the l=ltspice.Ltspice(filepath). However if the variables is less than 1000 it works.
I am wondering if the module has some limitation for handling the rawfile variables?

Please find the error message below for reference:

--> 441     l=ltspice.Ltspice(rawfile)
    442     l.parse()
    443     simtime = l.getTime()

C:\ProgramData\Anaconda3\lib\site-packages\ltspice\ltspice.py in __init__(self, file_path)
     43 
     44         self.header_size = 0
---> 45         self.read_header()
     46 
     47     def set_variable_dtype(self, t):

C:\ProgramData\Anaconda3\lib\site-packages\ltspice\ltspice.py in read_header(self)
     65             fp_line_end = 0
     66             while not ('Binary' in line or 'Values' in line):
---> 67                 if bytes([data[fp_line_end]]) == b'\n':
     68                     line = str(bytes(data[fp_line_begin:fp_line_end+2]), encoding='UTF16')
     69                     lines.append(line)

IndexError: index out of range

DongHoonPark, 2021-03-04:
Dear Goutham.

There is no programmatic limit for the number of variables.
As you succeed to parse your simulation result in small file, it may be
related to memory issue..
If you can share .raw file, I'll look into it.

Regards.

Goutham-AKA, 2021-03-04:
Dear Dong,

Thanks for your reply.

Sure, I can share the .raw file which I tried to run using ltspice module.

The attached zip folder consists of two raw files:
1. raw_file_5x5 --> succeed to parse the simulation 
2. raw_file_20x20 --> same file with higher matrix size fails to parse the simulation 

[raw_files.zip](https://github.com/DongHoonPark/ltspice_pytool/files/6083148/raw_files.zip)

DongHoonPark, 2021-03-04:
Dear Goutham.

I have figured out what the problem is.

My library has a function that automatically determines the encoding of a .raw file. 
Whether utf-8 or utf-16le.To do this, it scans the header (until variable declarations) and for the performance issue, it is limited to 100kB for default configuration.
Your second .raw file has the header which is over 100kB and it is the reason that the library throws an exception that is not intended.

The solution is simple. 

Modify the static variable of Ltspice class that indicated maximum header size.
I attach a code sample that allows 1MB of header size below.

```python
from ltspice import Ltspice

Ltspice.max_header_size = int(100e4)
lt = Ltspice("raw_file_20x20.raw")
lt.parse()
```

If you have any other problem, please let me know :D
Regards.

closed 2021-03-04 by DongHoonPark

Goutham-AKA, 2021-03-04:
Dear Dong,

Thanks for your response.
Indeed modifying the maximum header size in the raw file did the trick and the issue is solved.

Thanks & Regards,
Goutham