donghoonpark/ltspice_pytool #15: I have a problem with a .tran simulation rawfile, error raise FileSizeNotMatchException  ltspice.FileSizeNotMatchException ’

Wallflower123, 2021-03-09:
I have used the ltspice library recently to read rawfiles of both .tran and .ac simulation of the NgSpice simulator.
However, when it comes to this particular [one](https://drive.google.com/file/d/1W2zNAL5Iw6O-n9_yn6_to7VmtB1gT9W8/view), I get error  `raise FileSizeNotMatchException ltspice.FileSizeNotMatchException`.

I'm thinking maybe because it is voluminous(?)

Any ideas how to solve it? 

Thanks in advance

DongHoonPark, 2021-03-11:
Your .raw file uses a double-precision data type. 
So, you need to specify it before parsing.

```
from ltspice import Ltspice

l = Ltspice("your-raw-file.raw").set_variable_dtype(float)
l.parse()
```

Regards.

Wallflower123, 2021-03-15:
Have you tried it on the file I shared? 
Because I still get the same error unfortunately ...

DongHoonPark, 2021-03-15:
Yes, I did. I tried with the .raw file you have given which is a size 1.23GB. You'd better check the library version or code.

Regards.

Wallflower123, 2021-03-16:
You are right, I confused the libraries, how stupid of me!

Thank you so much for your time, your precious help and, guidance!

closed 2021-03-16 by DongHoonPark