donghoonpark/ltspice_pytool #16: Trouble when reading RAW file generated by DC analysis

aoki-works, 2021-03-26:
Hello DongHoonPark,
I met a problem reading the RAW file generated by DC analysis.
For example, if the voltage is swept from -50V to 0V as shown below, 
the voltage will change from 0V to -50V at the boundary of CASE.
```
.dc Vce 0 -50.0 0.01
.step param Ib list 1m 2m 10m
```
However, the code on line 206 could not detect this change.
```
205  for i in range(self._point_num - 1):
206      if self.time_raw[i] > self.time_raw[i + 1] and self.time_raw[i + 1] == start_value:
207          self._case_split_point.append(i + 1)
```
As a workaround, I modified the codes as follows and use it.
```
205  for i in range(self._point_num - 1):
206      if self.time_raw[i] > self.time_raw[i + 1] and self.time_raw[i + 1] == start_value:
207          self._case_split_point.append(i + 1)
206      elif self.time_raw[i] < self.time_raw[i + 1] and self.time_raw[i + 1] == start_value:
207          self._case_split_point.append(i + 1)
```
I hope this issue will be officially resolved.
best regards,

Aoki

DongHoonPark, 2021-03-26:
Oh.. It is totally unexpected situation because I thought the x-axis of simulation always rises (e.g. time). 
The issue may be cleared with the code below but I need time to test it in various conditions.

```python
for i in range(self._point_num - 1):
    if self.time_raw[i + 1] == start_value:
        self._case_split_point.append(i + 1)
```

Of course, I will handle this issue officially and make commit after unit tests.
Thank you for your dedicated report and please wait a moment to get the version of which the issue solved.

Regards.

DongHoonPark, 2021-07-24:
Hi @aoki-works 

Thank you for long waiting.
This issue resolved in [ltspice 1.0.4](https://pypi.org/project/ltspice/1.0.4/)

Regards.

closed 2021-07-24 by DongHoonPark