donghoonpark/ltspice_pytool #19: Error reading .raw file

nicolasdavister, 2021-06-23:
File ".../anaconda3/lib/python3.7/site-packages/ltspice/ltspice.py", line 45, in __init__
    self.read_header()
  File ".../anaconda3/lib/python3.7/site-packages/ltspice/ltspice.py", line 67, in read_header
    if bytes([data[fp_line_end]]) == b'\n':
IndexError: index out of range

DongHoonPark, 2021-06-30:
Hello Nicolas.
If you can provide me that .raw file, please send me to check it out.
You can send it to my mail or attach network drive link.
As I'm using my library to analyze simulation result which is larger than 5GB, I think it would not size-related issue.

Regards.

nicolasdavister, 2021-06-30:
Hello,

Here you are the link to the .raw file that causes the trouble.
https://www.transfernow.net/dltransfer/?utm_source=20210623YiwJwui2&utm_medium=&utm_content=fr <https://www.transfernow.net/dltransfer/?utm_source=20210623YiwJwui2&utm_medium=&utm_content=fr>
Thank you for the follow-up.

Regards,

Nicolas Davister

DongHoonPark, 2021-07-02:
I check out link but it already had been expired.
Can you give me link again?

nicolasdavister, 2021-07-02:
Of course, this is the newly generated link : https://www.transfernow.net/dl/20210702hqA5C402 <https://www.transfernow.net/dl/20210702hqA5C402>

Regards,

Nicolas Davister

Pankwings, 2021-07-03:
Hi Nicolas,

If you make one line change in function 'def read_header(self):' in ltspice.py file than the "IndexError: index out of range" error will be solved.
I have pasted the function on which I have made the changes here:
`    def read_header(self):
        filesize = os.stat(self.file_path).st_size
        print("\nfilesize :",filesize,"\nself.max_header_size :", self.max_header_size)
        with open(self.file_path, 'rb') as f:
            if filesize > self.max_header_size:
                #data = f.read(self.max_header_size)  
                data = f.read(filesize)
            else:
                data = f.read()  `

Please check whether it works for you.
Regards,

Pankwings

DongHoonPark, 2021-07-04:
Hi Nicolas,
I'm sorry for late response and thanks for walk around from @Pankwings 
But there is more simple trick. 
Actually, the real problem was, really long header which is derived from many variables.
So, what you only need to do is just extend the variable that indicates header size.
Like below.

```python
from ltspice import Ltspice
Ltspice.max_header_size = int(1e6)
l = Ltspice("TransfoPublicationReduced_OnlySecondary_TurnLevel.raw")
```

By this trick, you should get result object with no problem.
As it is the problem that happens frequently so I may have to consider more smarter way to read header with fast and safe..
If anyone have brilliant idea, please let me know to improve this lib.

Regards.

closed 2021-07-05 by DongHoonPark