donghoonpark/ltspice_pytool #21: Getting value of current .step

Cyberduke01, 2021-07-23:
Hi, apologies for making an issue, but I have a question. 

I have a simulation which implements a .step command for different values. 

The problem is that when I plot the values in python from the raw file, even though I can get the values for each case, I would like to know how I can get the values of the current case in code.  i.e. if I step Rs from 1 to 10 Ohm, how do I know what was the value for Rs in the current case?

DongHoonPark, 2021-07-24:
Hi @Cyberduke01 
Unfortunately, there is no direct way to know that current iteration value of simulation.
Because this lib only get the value of simulation result not input or condition.

However, there is simple walk-around that I frequently use.
Just make bv source which indicates the step condition and read it on library.

Here is simple example that you can try.
[image]

I had considered about to read .asc file to get step condition and it is not technically difficult, but there is danger to use schematic (.asc) and result (.raw) pair which is not match.
So I won't add that function in this library.

Regards.

Cyberduke01, 2021-07-26:
Hi, that is a good idea. Thank you for the answer

closed 2021-07-26 by DongHoonPark