donghoonpark/ltspice_pytool #32: Pandas support?

MarcelRobitaille, 2023-01-26:
Thanks for the fantastic library :smile:

Would you consider adding Pandas support? I am analyzing a tricky circuit, and I like using Pandas for this kind of task (smoothing, filtering, arithmetic between columns, etc.).

I am thinking of adding a method like `l.to_pandas()`. The implementation would be quite simple. This is working for me:
```python
def to_pandas(self):
    return pd.DataFrame(self.y_raw, columns=self._variables)
```

The reason I am asking and not just doing the above line in my code is that it uses a private attribute `_variables`. I am also not sure if `y_raw` is supposed to be public.

I can do a pull request with an example and tests if this is a feature you would include.

DongHoonPark, 2023-01-26:
I appreciate for your thoughtful idea and pandas support could be useful for some situations.
My original intention that had made `y_raw` as public variable was similar with your code.
Moreover, in the same sense, I made `variable` as accessible class property so that it can be used anywhere else.
You may try a function which using no private variable like below.
```python
def dataframe(variable_names:List[str], raw:List[Any]):
    return pd.DataFrame(raw, columns=variable_names)

def dataframe(lt:Ltspice):
    return pd.Dataframe(lt.raw, columns=lt.variables)
```
Regards.

MarcelRobitaille, 2023-01-26:
Sorry, I didn't notice the public property.

closed 2023-02-04 by DongHoonPark