donghoonpark/ltspice_pytool #33: Receiving "None" instead of a voltage in voltage source

cuevas46, 2023-02-22:
Dear Donghoon Park,

Thank you for creating this tool, it has been advantageous. 

I have two voltage sources in a circuit, and one of them (V1) is doing a sweep voltage from 0 to 1 in 0.01 steps while holding the other (V2) at 1V. I am doing some calculations where I need the value of V2 for each step in V1, meaning that I want to have a list of "1" values if I use get_data('V2'), instead what I get is "None". Any idea why this could be happening or how to solve it?

Best regards,

DongHoonPark, 2023-03-01:
You may need to check the name of voltage node. Net name to investigate voltage usually labeled as n001 or n002 unless you do not made net label. Please check the name of the voltage node you want to watch in the plot of ltspice and try to extract it with node name.

cuevas46, 2023-03-01:
Thank you, this solve the issue

closed 2023-03-12 by DongHoonPark